PaulStoffregen/Ethernet #4: Sockets do not close properly, they get stuck sometimes

htomi95, 2016-03-20:
On my Teensy 3.2 + Arduino Ethernet Shield I have a webserver. I need to send requests to it every half second. At some point I noticed that the webserver had stopped responding, it did not process any more requests. The Teensy or the Ethernet Shield had not frozen, they were still working properly - it also had a valid IP address. This happened randomly though - sometimes it worked for hours, sometimes it did not.

After some digging and debugging I reailzed that` EthernetServer.available()` returned zeros so the line following (the if state) returned false. Examining the `available()` function in `EthernetServer.cpp` I came to the conclusion that after some time all four sockets got used up even after closing the connection. I printed out the result of `socketRecvAvailable(i)` and four zeros were returned.

My code in the main loop looks as follows:
```
EthernetClient client = server.available();
if (client)
{
	String command = "";
	String getParams = "";

	while (client.connected())
	{
		if (client.available())
		{
			char c = client.read();
			getParams += c;

			if (c == '\n')
			{
				break;
			}
		}
		else
		{
			break;
		}
	}
	command = getCommand(getParams);
	if (!command.equals(""))
	{
		processCommand(command); // A function to process the raw GET params form HTTP header
	}
	client.println("HTTP/1.1 200 OK");
	client.println("Access-Control-Allow-Origin: *");
	client.println("Content-Type: application/json");
	client.println("Connection: close");
	client.println("Server: Teensy");
	client.println();
	printJsonResponse(&client, command.equals("") ? 0 : 1); // A function to print out JSON data
	delay(1);
	client.stop();
```
Even though the connection was closed after each request, after some time all four sockets were used up. I managed to solve the problem by adding a new function to `EthernetClient ` and calling it in the main loop which closes all sockets that return zero.
```
void EthernetClient::closeGhostSockets()
{
	for (int i = 0; i < MAX_SOCK_NUM; i++)
	{
		uint8_t stat = socketStatus(i);
		if (stat == SnSR::ESTABLISHED || stat == SnSR::CLOSE_WAIT)
		{
			if (socketRecvAvailable(i) == 0)
			{
				socketClose(i);
			}
		}
	}
}
```

yoonghm, 2017-07-20:
I guess the remote clients have closed connection but the library is not aware. Perhaps one can use tcp_keepalive_time to track inactivity and reset / close from the server when the time exceeded.

PaulStoffregen, 2017-10-26:
Sorry about the long delay.  Recently I've been looking into these long-standing Ethernet issues.

This change is meant to fix the problems where unexpected closed connections aren't always reclaimed.

https://github.com/PaulStoffregen/Ethernet/commit/7c452a96bb4c186239d28754cef40d86a5a973f9

PaulStoffregen, 2018-07-17:
Pretty sure the remote disconnect handling has fixed this issue.  Going to close.  If anyone still sees this problem, please reopen or start a new issue.

closed 2018-07-17 by PaulStoffregen