PaulStoffregen/Ethernet #6: How

czesiu, 2017-01-06:
Hi,

I'm tested your library on Controllino MEGA (based on MEGA 2560 with W5100 chip) and transfer is now about 6 times faster than with original Ethernet library - now it is about 27,2 kbytes / s.

I have couple questions:
1. What means columns Internet and Local in readme?
This is speed in kilobytes / s? What is difference between this columns?
2. Could you share code how you test that speed?
3. What means value "fail"?
No connectivity? Or compile error? Something else?
I ask this becauseI have MEGA 2560 and it works for me.

Thanks in advance

PaulStoffregen, 2017-01-06:
The test is just the WebClient example, with line 87 commented out (see the comment on that line).

https://github.com/PaulStoffregen/Ethernet/blob/master/examples/WebClient/WebClient.ino#L87

"Internet" means run it as-is.  "Local" means set up an Apache2 web server on the same Ethernet using a fast computer.  Edit the example to your local server's IP number and a web page you put on that server.

Sorry, I can't spend more time answering so many questions.  I'm glad you're using the library and it works well for you.  Please understand this is free software, which does not come with technical support.

czesiu, 2017-01-07:
Thanks

I tested using this example on Controllino MEGA (based on MEGA 2560 with W5100 chip) and I had following results:
Internet: 17,61 kbytes / s
Local: 19,76 kbytes / s

If you want to you can add this results to table or I can make PR with this.

PaulStoffregen, 2017-01-07:
Not yet.  I am going to do another round of optimizations and retest, so there's not much point to adding another entry to the table that's going to soon be replaced.

PaulStoffregen, 2018-07-17:
Doing more testing.  For now, the table is only going be tests I have personally run.

closed 2018-07-17 by PaulStoffregen