PaulStoffregen/Ethernet #9: Removal of reset signal code is causing support requests

macaba, 2017-02-07:
Following commit https://github.com/PaulStoffregen/Ethernet/commit/ee1eacf88b4c7d5e0b0a7f7ea1c94cc8ce8de7e1 the lack of reset signal has caused many forum threads:

https://forum.pjrc.com/threads/38780-wiz820io-server-not-initializing-to-fixed-IP
https://forum.pjrc.com/threads/41958-Ethernet-begin-hangs-after-power-cyle-in-Nov-29-2016-release
https://forum.pjrc.com/threads/42008-ethernet-init-issues-3-2-and-3-6

The exact code that was removed (and lack thereof causes this issue):

```
#define W5200_RESET_PIN  9

#ifdef W5200_RESET_PIN
  pinMode(W5200_RESET_PIN, OUTPUT);
  digitalWrite(W5200_RESET_PIN, LOW);
  delay(1);
  digitalWrite(W5200_RESET_PIN, HIGH);
  delay(150);
#endif
```

I've suggested a possible compromise in pull request https://github.com/PaulStoffregen/Ethernet/pull/8 but it still requires the user to know that they would need to add `#define WIZ_RESET`.

I'm not sure what the correct answer is, I defer to your expertise.

macaba, 2017-02-07:
Another thread:
https://forum.pjrc.com/threads/41936-Teensy-3-6-resetting-with-Serial-Wiz820-Ethernet?p=133034

PaulStoffregen, 2018-07-17:
I'm going to consider this problem "solved" by the addition of reset hardware on the adaptor boards and the fact that W5200 has been replaced by W5500 which has much better startup behavior.

closed 2018-07-17 by PaulStoffregen