PaulStoffregen/Ethernet #12: Support for Arduino Mega

bkkrunal, 2017-04-21:
Hi Paul,
I have been following you from last 6 months after joining Arduino developer's page and I would like to thank you for all your contribution.
I have been also experiencing very slow speed while using Arduino Ethernet shield with its original library, today I found your work here to improve the speed and would love to try it out. But in Readme.md, you have mentioned "FAIL"for Arduino Mega. Can you please suggest...

Thanks,
Krunal

PaulStoffregen, 2017-04-21:
I do not know.  It did not seem to work with the Mega clone I have.  I do not have a genuine Mega.  I am currently working on other projects, so I can't help with this.  Please give it a try on your Mega (and also on Uno or Teensy or another known good board) and then you tell me if it works!

bkkrunal, 2017-04-22:
Hi,
I have tested the library with my custom designed board running with ATmega328 at 16MHz, I used board name as "Arduino Pro or Pro mini, ATmega328(5V, 16MHz) for compiling and its working perfectly. Previously the response was very slow with Arduino's inbuilt libraries, but now I have observed great increase in response time.
This system is developed to control VFD frequencies and uses UDP pakets to communicate with the UDP GUI on Windows.
I am more than happy to tell you that previously with Arduino's default ethernet libraries when I was running this system with continuous UDP requests, it would work for 1 -2 hours and after that it would respond very slowly. Sometimes it would take more than 3 seconds by the device to respond to the GUI. But with your library, I have not experienced any lag till now even after running for more than 2 hours.
I will keep this device running for few days will be monitoring it the performance.

I tested the library with Arduino UNO (clone) and it worked very well.
I also tested it with Arduino Mega (clone). Initially sometimes it worked and sometimes didn't connect to internet with USB power supply, but when I connected external 12V DC adapter, it worked almost everytime.

I tested above functionalities for UNO and Mega with WebClient sketch.
I have tested them with W5100 ethernet shield. Also I have already ordered W5500 shield and will be testing that once I receive it.

I would also love to test the library with other boards, but currently I only have these with me. I also have stm32duino and will test that also later and update here.

I would also love to try Teensy and planning to buy one.
Thanks again for all your work.

Regards,
Krunal

PaulStoffregen, 2017-04-22:
> previously with Arduino's default ethernet libraries when I was running this system with continuous UDP requests, it would work for 1 -2 hours and after that it would respond very slowly.

Can you provide a test program and whatever other UDP-based software or service is needed elsewhere on the network, so I can run this test on boards here?

PaulStoffregen, 2017-04-22:
Several people have reported this same bug, slow or disconnect after a matter of hours.  So far, only one posted any code which wasn't complete.  I have run tests here on various boards, but never managed to reproduce this problem here.  It will never get solved, at least by my hand, until someone steps up with a program and reliable steps to reproduce the problem.

bkkrunal, 2017-04-23:
Yes, Of course.
I have documented here:
https://hackaday.io/project/12990-pressure-regulatory-system-for-solar-plant

The files are:
https://github.com/bkkrunal/VFD-Controller/tree/master/ArduinoVFDController

I am not using standard Arduino hardware, instead I have designed my own boards and compiled as Arduino Pro or Pro mini at 5V 16MHz. I hope it will work with standard hardware also.
It uses Interrupt to read UDP packets. So we need to short the INT jumper on ethernet shield.

But, this is not the latest work, but it was also having the same bug of getting slower after few hours. I will share the latest work soon after cleaning the final code.

I would be happy to get it work for you, Let me know if you face any problem.

PaulStoffregen, 2017-04-23:
I really need code that has been tested to reproduce the problem running on a standard Arduino and shield, ideally without any other hardware required.

When I try to reproduce bugs that occur with custom boards and hardware I don't have, the result is almost always a lot of wasted time not actually reproducing the bug.

bkkrunal, 2017-04-24:
OK, 
Give me some time. I will reproduce the bug with standard Arduino UNO/MEGA and will send you ready to use files.

PaulStoffregen, 2017-10-25:
Fixed.
https://github.com/PaulStoffregen/Ethernet/commit/a42c73aa6d501eda9abef3683fff80e4536d725a

closed 2017-10-25 by PaulStoffregen