PaulStoffregen/Ethernet #14: UDP on W5100 not realible

uzi18, 2017-08-07:
Hi Paul
This is bug in both current Arduino Ethernet and Your library, udp does not work after 2-3 requests for ntp time on W5100. 

For tests have used udpNTPClient example  with original EthernetShield (W5100) and W5200 shield on ArduinoMega with Your git master lib.
On W5100 problem exists, when changed shield to W5200, requests works ok and every ntp.
So it is W5100 code related.

From 2nd/3rd request udp stops working, did some debug here and it looks like not DNS realted,
have stored IPadress of timeserver and used method with IPAdress.

Any hint?

PaulStoffregen, 2018-07-17:
Please let me know if you find any way to reproduce the problem with the latest code.

Soon this version will become the official Arduino Ethernet lib, so I'd really like to resolve any outstanding issues before we publish it to so many people!

uzi18, 2018-07-18:
If it goes to official, it could be nice to have async dhcp and dns here, what do you think?