PaulStoffregen/Ethernet #19: Arduino M0 Support

itsnotyasser, 2018-03-21:
Hi, first of all thank you for your branch of the Ethernet library.
I've been trying to use the WIZ820IO ( Using the W5200 controller) with an Arduino M0 Pro, looking at the list of hardware supported by the branch, it seems it does support the Arduino Zero so I was wondering if there was support for the M0 and how to get it working.

PaulStoffregen, 2018-03-21:
Have you tried it?  Can you confirm M0 Pro in fact does not work?

itsnotyasser, 2018-03-21:
I've just triple checked all connections between my WIZ820io and arduino M0 Pro using the SPI from the ICSP header. I ran the simple DHCP Address printer from your example code with the slight modification of replacing Serial.println with SerialUSB.println ( Arduino m0 Native USB). And the sketch does not get an IP from DHCP.

uzi18, 2018-03-21:
First check if begin()  detects wiznet chip.

PaulStoffregen, 2018-03-21:
Do any of the other libraries work?  Maybe the one from Wiznet?  If your hardware really is connected correctly, wouldn't that one at least work?

itsnotyasser, 2018-03-21:
@uzi18 , sorry I'm quite the beginner. How would I do that?
@PaulStoffregen I've tried with multiple libraries, most did not compile because the SPI Class was different. I did not find any Ethernet library from Wiznet, do you have more information?

PaulStoffregen, 2018-03-21:
Your best bet would be to ask on the Arduino forum.  Maybe others who've actually used M0 Pro will see your message and reply?

https://forum.arduino.cc/index.php?board=98.0

When you post on the forum, I highly recommend a little extra effort to share photos of your wiring, links to exactly which parts you're using, posting the program you're running to test, and so on.  Yes, this takes extra work.  But consider *many* people post questions to the forum with very little info.  People on forums like to help those to invest a little extra work, and usually aren't helpful to people who appear to be lazy.  So put a little work into your forum post to give complete info....

itsnotyasser, 2018-03-21:
Thank you, I'll give it a go

PaulStoffregen, 2018-03-21:
As far as M0 Pro goes, as nearly as I can tell the board has been discontinued.

https://store.arduino.cc/usa/arduino-m0-pro

Since it's so similar to Zero, I believe things should "just work", as long as you avoid pins 2 & 4 which aren't used for SPI anyway.  But if it doesn't work, I'm afraid I can't do much to help.  I don't have a M0 Pro, and since it's discontinued, I don't see myself going to a lot of work to obtain one.

But I do have Arduino Zero in the bin of Arduino boards I use for testing libraries.

itsnotyasser, 2018-04-27:
Hi @PaulStoffregen, just an update.

After extensively trying with a lot of libraries and trying to rewrite the W5200 files to work with my M0 it seems there are some compatibility issues. 
I tried with the M0 board and a ETH WIZ click ethernet adapter ( equipped with the W5500 Ethernet controller) and your library just works.

Thanks for the help!

PaulStoffregen, 2018-07-17:
I tested just now on a MKR1000 using the WIZ820io module.  Works fine.

[image]

PaulStoffregen, 2018-07-17:
Tested Arduino Zero with WIZ820io.  Also works.

[image]

PaulStoffregen, 2018-07-17:
I don't have the M0 Pro board, but since it's working on the very similar Arduino Zero and MRK1000 boards, and so far nobody's replied (no news is good news), I'm going to close this issue.

closed 2018-07-17 by PaulStoffregen