PaulStoffregen/Ethernet #27: Support for Controllino

TimoErdmann, 2018-07-27:
Hey, 

is it possible to add support for the Controllino Hardware? https://controllino.biz
The SS pin for the W5100 is on Port: PJ3 (you can read this here: https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library)

I tried the current version of this library, but it doesn't work. I believe it uses the wrong SS Pin? The default is pin 10 as I found out.

Moreover, is there a possibility to reset the Ethernet Chip? Sometimes it hangs with the old library :( 

Greetings 
Timo

PaulStoffregen, 2018-07-27:
Let me answer a question with another question...

Which digital pin number is used for PJ3?

PaulStoffregen, 2018-07-27:
If PJ3 does not have a digital pin number assigned (as it appeared when I looked at Controllino months ago), then *that* must be fixed in Controllino board support package.

Then just use Ethernet.init(pin) before Ethernet.begin().  Very easy.  But Controllino **must** have a digital pin number of PJ3.  If it does not, then Controllino is incompatible and must be fixed.

PaulStoffregen, 2018-08-11:
I've created an issue with Controllino to ask them to update their board support library.

https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/issues/20

PaulStoffregen, 2018-08-11:
I'm going to close this issue, since I've contributed this stuff back to Arduino's library for all boards.

closed 2018-08-11 by PaulStoffregen