PaulStoffregen/Ethernet #28: SPI clock on Arduino Pro Mini (3.3V, 8 MHz) ATmega328P

TomMajor, 2018-08-19:
Hi,
I want to use the Improved Performance Ethernet Library on this device:
Arduino Pro Mini (3.3V, 8 MHz) ATmega328P
for example this one
https://www.adafruit.com/product/2377
(many clones out there).
(Maximum ethernet throughput is not needed on the project)

I want to use a W5500 / wiz850io (ordered, but not arrived yet).

You have this code in the library
`// Safe for all chips`
`#define SPI_ETHERNET_SETTINGS SPISettings(14000000, MSBFIRST, SPI_MODE0)`

On a 8 MHz device, the max. SPI clock rate would be 4MHz. As far as I can see, original Arduino code in SPI.h handles this correctly by limiting any higher clock demand to fosc/2:
`      if (clock >= F_CPU / 2) {
        clockDiv = 0;
...
}
`

My first question: Will a 4MHz SPI clock run fine with this library or is there a limit/problem somewhere if the SPI clock gets 'too slow' ?

Second question, should I connect the /RST pin of the wiz850io to Arduino Reset, or doesn't it really matter (because it get initialized by SW anyway)?

Thanks for your work and a great library,

PaulStoffregen, 2018-08-19:
Github issues are not the place for questions.  Ask on the Arduino forum.

closed 2018-08-19 by PaulStoffregen

TomMajor, 2018-08-20:
That's ok. I thought maybe missing documentation about suitability of this lib for 'slow' AVR devices / SPI clocks would be something that could be added in the readme or header file if it is known by the author.
I will test it and ask further questions on the Arduino forum if problems arise, thanks.