PaulStoffregen/Ethernet #34: Ethernet::begin incompatible with ESP8266

GiovanniCmpaner, 2019-05-27:
Currently using 
```void EthernetClass::begin(uint8_t *mac, IPAddress ip, IPAddress dns, IPAddress gateway, IPAddress subnet)```
generates a compiler error, as IPAddress from the ESP8266 core doesn't have a `_address` member:
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/IPAddress.h

These lines:
https://github.com/PaulStoffregen/Ethernet/blob/dbed40a1a308855497dfe3f7a8d8fb5cfdcce2e5/src/Ethernet.cpp#L86-L94

Should be:
```
W5100.setIPAddress(ip.raw_address());
W5100.setGatewayIp(gateway.raw_address());
W5100.setSubnetMask(subnet.raw_address());
```
It isn't necessary to check the version, as raw_address() is available since 1.0.5:
https://github.com/arduino/Arduino/blob/22997bbfd5c841a60e9a5909868c62cd459c798d/hardware/arduino/cores/arduino/IPAddress.h

michaelradhuber, 2019-10-05:
Thanks @GiovanniCmpaner , I would like to confirm that. @PaulStoffregen could you plz implement those changes? Thank you very much! Great work!

PaulStoffregen, 2019-10-05:
Which boards have you tested with this change?

Looks like the raw_address() function is private on Arduino's core library.

GiovanniCmpaner, 2019-10-05:
Any board that uses ESP8266 Arduino Core.
As you can see in the source: https://github.com/esp8266/Arduino/blob/master/cores/esp8266/IPAddress.h
The class doesn't have a _address variable.