PaulStoffregen/Ethernet #48: ESP32 crashes on Ethernet.begin

Zambiorix, 2020-07-29:
Every now and again I get a crash on:

`
Ethernet.begin
`

I run the connect/disconnect code in a separate task.
What could be wrong?

Thanks

`
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400db145  PS      : 0x00060630  A0      : 0x800db7db  A1      : 0x3ffb5690  
A2      : 0x0000bdd5  A3      : 0x3ffc9d8c  A4      : 0x4001846c  A5      : 0x00000020  
A6      : 0x0000004c  A7      : 0x00000000  A8      : 0x000000d5  A9      : 0x3ffb5660  
A10     : 0x00000002  A11     : 0x3ffbee76  A12     : 0x3ffc9cdc  A13     : 0x000003a0  
A14     : 0x00000034  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40018470  LBEG    : 0x4000c428  LEND    : 0x4000c42d  LCOUNT  : 0x00000000  

Backtrace: 0x400db145:0x3ffb5690 0x400db7d8:0x3ffb56c0 0x400dad0f:0x3ffb56f0 0x400dc29a:0x3ffb5710 0x400dc76b:0x3ffb5770 0x400dc8a9:0x3ffb57a0 0x400daa19:0x3ffb57c0 0x400d4961:0x3ffb57f0 0x400d4550:0x3ffb5820 0x400d45b1:0x3ffb5960 0x400d45bb:0x3ffb5980 0x40089629:0x3ffb59a0
  #0  0x400db145:0x3ffb5690 in getSnTX_FSR(unsigned char) at .pio/libdeps/esp32doit-devkit-v1/Ethernet/src/socket.cpp:481
  #1  0x400db7d8:0x3ffb56c0 in EthernetClass::socketBufferData(unsigned char, unsigned short, unsigned char const*, unsigned short) at .pio/libdeps/esp32doit-devkit-v1/Ethernet/src/socket.cpp:488
  #2  0x400dad0f:0x3ffb56f0 in EthernetUDP::write(unsigned char const*, unsigned int) at .pio/libdeps/esp32doit-devkit-v1/Ethernet/src/EthernetUdp.cpp:94
  #3  0x400dc29a:0x3ffb5710 in DhcpClass::send_DHCP_MESSAGE(unsigned char, unsigned short) at .pio/libdeps/esp32doit-devkit-v1/Ethernet/src/Dhcp.cpp:169
  #4  0x400dc76b:0x3ffb5770 in DhcpClass::request_DHCP_lease() at .pio/libdeps/esp32doit-devkit-v1/Ethernet/src/Dhcp.cpp:56
  #5  0x400dc8a9:0x3ffb57a0 in DhcpClass::beginWithDHCP(unsigned char*, unsigned long, unsigned long) at .pio/libdeps/esp32doit-devkit-v1/Ethernet/src/Dhcp.cpp:23
  #6  0x400daa19:0x3ffb57c0 in EthernetClass::begin(unsigned char*, unsigned long, unsigned long) at .pio/libdeps/esp32doit-devkit-v1/Ethernet/src/Ethernet.cpp:230
  #7  0x400d4961:0x3ffb57f0 in dt::NetworkEthernet::task_event(bool&, unsigned int) at src/Network/ModuleEthernet.cpp:130
  #8  0x400d4550:0x3ffb5820 in dt::NetworkModuleBase::task_loop(dt::NetworkBase*, unsigned int, unsigned int, unsigned int) at src/Network/Module.h:155
  #9  0x400d45b1:0x3ffb5960 in dt::NetworkModuleBase::NetworkModuleBase(dt::NetworkBase*, char const*, unsigned int, unsigned int, unsigned int, unsigned int)::{lambda(void*)#1}::operator()(void*) const at src/Network/Module.h:215
  #10 0x400d45bb:0x3ffb5980 in dt::NetworkModuleBase::NetworkModuleBase(dt::NetworkBase*, char const*, unsigned int, unsigned int, unsigned int, unsigned int)::{lambda(void*)#1}::_FUN(void*) at src/Network/Module.h:216
  #11 0x40089629:0x3ffb59a0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...

`

PaulStoffregen, 2020-07-29:
I am no longer maintaining Ethernet for non-Teensy boards.  This repository is only for Teensy.  You're asking for ESP help in the wrong place.

closed 2020-07-29 by PaulStoffregen

Zambiorix, 2020-07-30:
Why don't you make a statement about this in the main README?
ESP32 is mentioned several times there... (and many other non-Teensy boards)

(BTW I'm not free-riding here, I have bought several Teensy boards in the past)

PaulStoffregen, 2020-07-30:
Ok, good point.  I've updated the readme file.  Hopefully the Teensy-only status of this repository will be clear in the future.