PaulStoffregen/Ethernet #59: Does this work on T4?

Olivierdemoustier, 2022-11-28:
I get compiler error if I use T4. T3 is working fine

PaulStoffregen, 2022-11-28:
Ask on the forum.

https://forum.pjrc.com/forums/3-Technical-Support-amp-Questions

When you post on the forum, show a screenshot or exact copy of the error.  Clearly state which version of the software and which OS you're using.

closed 2022-11-28 by PaulStoffregen

Olivierdemoustier, 2022-11-28:
Hi Paul, not used on posting items on a forum, 
This is the error I get when I compile for . If I select TT, it compiles perfectly.
"D:\OneDrive\Arduino\TMS\TMS sacn to neopixel4\w5100.cpp:13:21: fatal error: SPIFIFO.h: No such file or directory
compilation terminated.
Multiple libraries were found for "Adafruit_NeoPixel.h"
  Used: D:\OneDrive\Arduino\libraries\Adafruit_NeoPixel
  Not used: C:\Users\olivi\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.57.1\libraries\Adafruit_NeoPixel
exit status 1

Compilation error: SPIFIFO.h: No such file or directory"

Olivierdemoustier, 2022-11-28:
Sorry, forgot to mention . I'm using windows  and Arduino IDE2.02

PaulStoffregen, 2022-11-28:
This is not a library defect.

PaulStoffregen, 2022-11-28:
Tech support is provided on the forum, **not here**.

Use github issues only for confirmed software defects.

PaulStoffregen, 2022-11-28:
If you reply here, make your reply *only* a link to the forum thread.  Do not continue this conversation here!